vectr-ucla/direct_lidar_inertial_odometry #1: positive constants in geometric observer

AFEICHINA, 2023-06-10:
Hello author,
    I am delighted to see that you have open-sourced this work. I tested some data right away and it worked very well. I am interested in the positive constants in the geometric observer. How is it set up? Does it relate to the usage scenario?
    Thanks again for your outsanding contribution!

kennyjchen, 2023-06-13:
Hi @AFEICHINA -- I'm glad to hear DLIO is working well for you. Yeah we have an internal procedure to retrieve gains for our geometric observer, which will be discussed in the next iteration of the [paper](https://arxiv.org/abs/2303.02777). Thanks for the interest.

AFEICHINA, 2023-06-14:
> 嗨 - 我很高兴听到DLIO对您来说效果很好。是的，我们有一个内部程序来检索几何观察器的增益，这将在[本文](https://arxiv.org/abs/2303.02777)的下一次迭代中讨论。感谢您的关注。

Thanks for your reply, I will close this issue.

AFEICHINA, 2023-06-14:
> Hi @AFEICHINA -- I'm glad to hear DLIO is working well for you. Yeah we have an internal procedure to retrieve gains for our geometric observer, which will be discussed in the next iteration of the [paper](https://arxiv.org/abs/2303.02777). Thanks for the interest.

Thanks for your reply, I will close this issue.