vectr-ucla/direct_lidar_inertial_odometry #2: Point Cloud Visualizations

rafaelspring, 2023-06-10:
Just a quick question: which app/lib/package was used to create the point cloud visualizations? These look very nice.

kennyjchen, 2023-06-13:
Hi @rafaelspring -- we just use [RViz](http://wiki.ros.org/rviz/DisplayTypes/PointCloud) with a large decay time and low alpha value on the motion-corrected point clouds.

juliangaal, 2023-06-15:
I guess you're talking about the setting in [this file](https://github.com/vectr-ucla/direct_lidar_inertial_odometry/blob/master/launch/dlio.rviz)?

rafaelspring, 2023-06-15:
@kennyjchen thanks for the pointer! Are you using a particularly high resolution to render these or does RViz already give you the nice transparency effects using a low alpha value?

kennyjchen, 2023-06-15:
It's a combination of RViz + a low alpha value and also using the full, motion-corrected point cloud during rendering.

closed 2023-06-16 by kennyjchen