vectr-ucla/direct_lidar_inertial_odometry #4: Livox+px4

Joosoo1, 2023-06-16:
Hello author, thank you for your open source. Can DLIO be implemented using livox+px4?

kennyjchen, 2023-06-16:
Hi @Joosoo1 -- yes, try using the [`feature/livox-support`](https://github.com/vectr-ucla/direct_lidar_inertial_odometry/tree/feature/livox-support) branch! I haven't tested it out with real data though, so let me know if it works or not.

Joosoo1, 2023-06-18:
Yes, thanks, I'll try it soon.

Joosoo1, 2023-06-18:
Well, I just tested DLIO on a livox mid-360 (built-in imu) and it doesn't seem to work, but it doesn't show any errors, then I tested DLO and DLO works fine, so I don't know where the error is happening, but it shouldn't be the reason for the timestamp being out of sync, because the mid-360 should be in sync. I don't know where the error occurred, but it shouldn't be the reason for the timestamp being out of sync, because mid-360 should be in sync! As you can see below, there is no next step! 
+-------------------------------------------------------------------+ | Direct LiDAR-Inertial Odometry v1.2.0 | +---- -------------------- -------------------------------------------+

 Calibrate IMU for 3 seconds... Done

 Acceleration Deviation [xyz]: 0.01453409, -0.00895136, -8.81063747 
Gyroscope Deviation [xyz]: -0.00210290, 0.00037739, 0.00247880

 DLIO initialization!

kennyjchen, 2023-06-19:
Hmm, think you could upload a bag for me? I haven't tested that branch out since I don't have any Livox data, so I can take a look if you provide some data. Thank you.

Joosoo1, 2023-06-21:
ok,经过我的测试。livox-mid360可以成功运行DLIO，只需要把livox_ros_driver2中的点云发布形式“xfer_format”改为“2”，就可以成功运行DLIO，如下：
[image]
[image]

closed 2023-06-23 by Joosoo1