vectr-ucla/direct_lidar_inertial_odometry #5: the output has been unchanged

WeilongXia, 2023-06-17:
Thanks a lot for opening source such a great work, but when I run the code, I found that the odometry output pose has always been the origin and has not changed. I have checked the data topics of lidar and imu, and there is nothing wrong. The frequency of lidar is 10Hz, and the frequency of imu is 200Hz. What are the possible reasons for the above case? Looking forward to your reply~
[image]
[image]
[image]

kennyjchen, 2023-06-19:
Yeah I haven't tested Livox support out since I don't have any data. Think you could upload a bag for me to play around with? Thanks!

WeilongXia, 2023-06-20:
This is the dataset of Livox-mid360: https://1drv.ms/u/s!AjkIEg1kWUNeiAIZHIDBTzXmtHqf?e=cWek2u 
And the extrinsic between the Lidar frame and IMU frame is 
[image]

Joosoo1, 2023-06-20:
Yes, it seems that we are experiencing the same problem, may I ask if you are using it as a navigation for robots? Drones or unmanned vehicles?Maybe we can communicate that i have recently encountered some difficulties.

WeilongXia, 2023-06-20:
Yeah I 'm going to use it as a localization module for drones, and maybe we can use the g-mail to communicate: xiawl0524@gmail.com

> Yes, it seems that we are experiencing the same problem, may I ask if you are using it as a navigation for robots? Drones or unmanned vehicles?Maybe we can communicate that i have recently encountered some difficulties.↳

Joosoo1, 2023-06-21:
ok, after my test, livox-mid360 can run DLIO successfully, just change the point cloud publishing form "xfer_format" to "2" in livox_ros_driver2, then it can run DLIO successfully. DLIO can be run successfully, as follows:

[image]
[image]

Joosoo1, 2023-06-21:
But the results do not seem to be very good, odom data z-axis jitter will be relatively large, may be due to the reason of not tuning parameters, but also may be due to the built-in imu noise is too large, the follow-up we can exchange about how to make it better and more stable, in addition, I am also intended to apply it to the top of the drone, used as navigation and positioning.

WeilongXia, 2023-06-21:
Thanks to your reply, the dimension of the acceleration data from the build-in imu in Livox MID-360 is g(gravity) instead of m/s^2, maybe you should multiply the data with a constant value like 9.8?

Joosoo1, 2023-06-21:
> Thanks to your reply, the dimension of the acceleration data from the build-in imu in Livox MID-360 is g(gravity) instead of m/s^2, maybe you should multiply the data with a constant value like 9.8?

ok,i will try!If you have better results, please let me know too!

Joosoo1, 2023-06-21:
wow! Your results look great, the position estimation is accurate and the built point cloud map has no drift, but my results are as bad as ever, still with a large offset on the z-axis, can you share your configuration parameters file? Also, about the z-axis, my livox has a z-axis  accel bias of -5, while yours is 0. I think there is something wrong here.
[image]
[image]

kennyjchen, 2023-06-21:
Hi @WeilongXia and @Joosoo1 -- thanks for the great discussion you two have started here. Looks like the Livox branch is working! It just needs to be published in the standard format.

Yeah something is definitely wrong if the IMU biases are that large. Have you checked your extrinsics? That's typically an issue, especially since the convention for IMU and LiDAR are different.

kennyjchen, 2023-06-21:
@Joosoo1 Yes, change the units of the accelerometer to m/s^2 as per the advice above.

Could someone record me a dataset in the new xfer_format? Thanks.

Joosoo1, 2023-06-23:
Well, it's true that because of the acceleration units, it has to be m/s^2 for it to work well.Later, I will provide you with the dataset.
`void dlio::OdomNode::getScanFromROS(const sensor_msgs::PointCloud2ConstPtr& pc) {

  pcl::PointCloud<PointType>::Ptr original_scan_ (boost::make_shared<pcl::PointCloud<PointType>>());
  pcl::fromROSMsg(*pc, *original_scan_);

  // Remove NaNs
  std::vector<int> idx;
  original_scan_->is_dense = false;
  pcl::removeNaNFromPointCloud(*original_scan_, *original_scan_, idx);

  // Crop Box Filter
  this->crop.setInputCloud(original_scan_);
  this->crop.filter(*original_scan_);

  // automatically detect sensor type
  this->sensor = dlio::SensorType::UNKNOWN;
  for (auto &field : pc->fields) {
    if (field.name == "t") {
      this->sensor = dlio::SensorType::OUSTER;
      break;
    } else if (field.name == "time") {
      this->sensor = dlio::SensorType::VELODYNE;
      break;
    } else if (field.name == "offset_time") {
      this->sensor = dlio::SensorType::LIVOX;
      break;
    } else if (field.name == "timestamp") {
      this->sensor = dlio::SensorType::HESAI;
      break;
    }
  }

  if (this->sensor == dlio::SensorType::UNKNOWN) {
    this->deskew_ = false;
  }

  this->scan_header_stamp = pc->header.stamp;
  this->original_scan = original_scan_;

}
`
This distortion correction does not seem to work！

dxdydz3dt, 2023-06-23:
Can you translate in English Please!. I am trying to run Livox but its not working

Joosoo1, 2023-06-23:
> Can you translate in English Please!. I am trying to run Livox but its not working
You can take a look at this: https://github.com/vectr-ucla/direct_lidar_inertial_odometry/issues/5#issuecomment-1600450012

It is necessary to additionally convert the acceleration units of the /livox/imu topic output to m/s^2,then the DLIO will work well, of course all this needs to be under the source code of the livox branch.

kennyjchen, 2023-06-23:
Hmm, yeah unfortunately it looks like in the `livox_ros_driver2` package, only `x` `y` `z` and `intensity` fields are copied over when publishing using the `xfer_format: 2` value. 

https://github.com/Livox-SDK/livox_ros_driver2/blob/master/src/lddc.cpp#L437

It looks like with their custom message, they have [per-point timestamps ](https://github.com/Livox-SDK/livox_ros_driver2/blob/master/src/lddc.cpp#L292). However, it looks like both [`timestamp`](https://github.com/Livox-SDK/livox_ros_driver2/blob/master/src/lddc.cpp#L292) and [`offset_time`](https://github.com/Livox-SDK/livox_ros_driver2/blob/master/src/lddc.cpp#L292) are used, which is a bit confusing since [their CustomPoint.msg contains `offset_time`](https://github.com/Livox-SDK/livox_ros_driver2/blob/master/msg/CustomPoint.msg), but the data above (the first link) has a `timestamp` field.

If only timestamps are available using their custom point format, we'll need to build support for that (unless they add timestamps to the PCL format). I'll keep digging.

kennyjchen, 2023-06-24:
Ok -- I think I might have figured this out. For future reference, this is how different sensors timestamp their points:
- Ouster: field `t` of type `std::uint32_t`, which is the time relative to the beginning of the scan in nanoseconds
- Velodyne: field `time` of type `float`, which is the time relative to the beginning of the scan in seconds
- Livox: field `timestamp` of type `double`, which is the absolute time of the point since epoch in nanoseconds
- Hesai: field `timestamp` of type `double`, which is the absolute time of the point since epoch in seconds

I pushed a few changes to the `feature/livox-support` branch. Please test this out using the original `xfer_format` value so that there are timestamp fields.

@WeilongXia -- I played around with your data, and it looks like the LiDAR and IMU measurements are not time synchronized. Is there an option in the Livox driver to synchronize them somehow?

WeilongXia, 2023-06-24:
Ok, I found a timesync choice in the config file like this
[image]
and I'll try to record a new rosbag.

juliangaal, 2023-06-24:
@WeilongXia could you rename the issue to something more meaningful, if you don't mind, for future reference? Something like "No localization with Livox Lidar"

WeilongXia, 2023-06-25:
> 

I tested the 'feature/livox-support' branch with the original xfer_format, but there is no localization output.

yijing2050, 2023-06-26:
> ld multiply the data with a constant value like 9.8?

@Joosoo1 I used xfer_format  = 0,and there was no point cloud. My wechat:yijingxs, you can add me and it is convenient to communicate with each other

Joosoo1, 2023-06-26:
> 
Yes, I tested the new branch and the DLIO only has odom output when xfer_format=2, but the result is completely wrong, so I have been using the original DLIO now, the original DLIO works well, I tested it on the drone and the positioning is very good. But deskwing is not working and will have some effect on DLIO，So I look forward to the author's help in solving this problem.

Joosoo1, 2023-06-26:
> > ld multiply the data with a constant value like 9.8?
> 
> @Joosoo1 I used xfer_format = 0,and there was no point cloud. My wechat:yijingxs, you can add me and it is convenient to communicate with each other
xfer_format = 2, old DLIO could working!

Joosoo1, 2023-06-26:
> > > ld multiply the data with a constant value like 9.8?
> > 
> > 
> > @Joosoo1 I used xfer_format = 0,and there was no point cloud. My wechat:yijingxs, you can add me and it is convenient to communicate with each other
> > xfer_format = 2, old DLIO could working!

Also, when you use imu with acceleration units of m/s^2, you should change the https://github.com/vectr-ucla/direct_lidar_inertial_odometry/tree/feature/livox-support/src/ under the livox branch Lines 1373 to 1375 of the code in dlio/odom.cc should be changed to `Eigen::Vector3f lin_accel(imu_raw->linear_acceleration.x * gravity_, imu_raw->linear_acceleration.y * gravity_, imu_raw->linear_acceleration.z * gravity_);` recompile！

yijing2050, 2023-06-26:
Thanks for your reply and kind advice. I have another problem when compiling, the first is the version of cmake. As my ros version is melodic,and the cmake version is 3.10.2, so I modify 

#VERSION 3.12.4
cmake_minimum_required(VERSION 3.10.2)
project(direct_lidar_inertial_odometry)
Could this modification do influence the output of the code?

 it have no add_compile_definitions.Then I comment out the section:# Not all machines have <cpuid.h&gt; available
#set(CMAKE_TRY_COMPILE_TARGET_TYPE STATIC_LIBRARY)
#file(WRITE ${CMAKE_BINARY_DIR}/test_cpuid.cpp "#include <cpuid.h&gt;")
#try_compile(HAS_CPUID ${CMAKE_BINARY_DIR} ${CMAKE_BINARY_DIR}/test_cpuid.cpp)
#file(REMOVE ${CMAKE_BINARY_DIR}/test_cpuid.cpp)
#if(HAS_CPUID)
#  add_compile_definitions(HAS_CPUID)
#endif()
after done above, I compiled successfully.

    

  
   
Also, when you use imu with acceleration units of m/s^2, you should change the https://github.com/vectr-ucla/direct_lidar_inertial_odometry/tree/feature/livox-support/src/ under the livox branch Lines 1373 to 1375 of the code in dlio/odom.cc should be changed to Eigen::Vector3f lin_accel(imu_raw-&gt;linear_acceleration.x * gravity_, imu_raw-&gt;linear_acceleration.y * gravity_, imu_raw-&gt;linear_acceleration.z * gravity_); recompile！

Joosoo1, 2023-06-26:
> 

I do not know much about this problem, but it should not affect if the compilation is successful, you can actually test it!

Joosoo1, 2023-06-26:
我对这个问题不太了解，不过编译成功的话应该不会影响，可以实际测试一下！

> Thanks for your reply and kind advice. I have another problem when compiling, the first is the version of cmake. As my ros version is melodic,and the cmake version is 3.10.2, so I modify #VERSION 3.12.4 cmake_minimum_required(VERSION 3.10.2) project(direct_lidar_inertial_odometry) Could this modification do influence the output of the code?  it have no add_compile_definitions.Then I comment out the section:# Not all machines have <cpuid.h&gt; available #set(CMAKE_TRY_COMPILE_TARGET_TYPE STATIC_LIBRARY) #file(WRITE ${CMAKE_BINARY_DIR}/test_cpuid.cpp "#include <cpuid.h&gt;") #try_compile(HAS_CPUID ${CMAKE_BINARY_DIR} ${CMAKE_BINARY_DIR}/test_cpuid.cpp) #file(REMOVE ${CMAKE_BINARY_DIR}/test_cpuid.cpp) #if(HAS_CPUID) #  add_compile_definitions(HAS_CPUID) #endif() after done above, I compiled successfully.
> […](#)

I don't know much about this, but if it compiles successfully, you should be able to run the program successfully, so you can test it as soon as possible!

kennyjchen, 2023-06-26:
@yijing2050 Unfortunately DLIO only supports ROS Noetic, but it sounds like you got it working. Removing those lines won't have any affect on the output; it's just to build support for [ARM architectures](https://github.com/vectr-ucla/direct_lidar_odometry/issues/16).

@WeilongXia @Joosoo1 The `xfer_format` should be `0` so that it's a message of type PointCloud2 but contains the timestamps. I will look into this again later this week. Sorry to ask again -- could someone record me a new bag that ideally time synchronizes the Mid-360's LiDAR and IMU (if not already), using `xfer_format: 0`? Thank you. In the meanwhile, you can turn deskewing off, but note that (especially at 10Hz) the point clouds will be quite distorted.

yijing2050, 2023-06-27:
@kennyjchen 
[Uploading livox_repub.zip…]()

yijing2050, 2023-06-27:
Actually, I have compiled successfully although, but i haven't got the odom.. The configuration is as the below.
[image]
[image]
[image]
[image]
[image]
[image]

kennyjchen, 2023-06-27:
Ok -- after digging through the Livox driver and also looking at [this dataset](https://github.com/vectr-ucla/direct_lidar_inertial_odometry/issues/5#issuecomment-1598064297), it seems like there is a bug for per-point timestamps when using `xfer_format: 0`(i.e., `Livox pointcloud2(PointXYZRTLT) pointcloud format`). The point cloud and IMU timestamps look synchronized, but the point timestamps are off and do not fall within the sweep window for some reason.

I looked at other Mid-360 packages, and it seems like most people use `xfer_format: 2` so that the `offset_time` field contains the time since the beginning of the sweep.

So new plan: please try the [updated branch](https://github.com/vectr-ucla/direct_lidar_inertial_odometry/commit/6b152351357708dd5e5db2ccb0c53c65aad656ad) but with `xfer_format: 2`. Your topic of type `livox_ros_driver::CustomMsg` should be called `/robot/livox`, which will call [this new callback](https://github.com/vectr-ucla/direct_lidar_inertial_odometry/blob/feature/livox-support/src/dlio/odom.cc#L1000-L1023). This is all assuming that [`offset_time`](https://github.com/vectr-ucla/direct_lidar_inertial_odometry/blob/feature/livox-support/src/dlio/odom.cc#L1000-L1023) is in nanoseconds relative to the beginning of the sweep (which I think is the case).

(Please be careful with the units of gravity. I [removed](https://github.com/vectr-ucla/direct_lidar_inertial_odometry/commit/6b152351357708dd5e5db2ccb0c53c65aad656ad#diff-2ed26eca15f057967bfc05c426ba09f51e70c50d65116f355b109b2f84269acbL854-L856) the previous multiplication with gravity.)

Note that I have not tested this at all, because I do not have a Livox sensor or any bags with `xfer_format: 2`. If someone could record me one that would be great so I can test it on my side. Otherwise, let me know if it works or not!

Joosoo1, 2023-06-28:
Sorry for the late reply, and thank you  @kennyjchen author for your efforts to solve this problem. Here is the dataset I recorded: accel in units when g
https://onedrive.live.com/?cid=6FAEBC9C8F25AA2C&id=6FAEBC9C8F25AA2C%21120&parId=root&o=OneUp

kennyjchen, 2023-06-28:
Ok, I think I got it working. Credit to [this Tweet](https://twitter.com/devemin/status/1631461802084302850?lang=en) for supplying the data I played around with.

[image]

Please use the newer [`livox_ros_driver2` package](https://github.com/Livox-SDK/livox_ros_driver2) with `xfer_format: 1`. The latest `feature/livox-support` branch seems to be working with deskewing. Let me know.

Joosoo1, 2023-07-02:
> Ok, I think I got it working. Credit to [this Tweet](https://twitter.com/devemin/status/1631461802084302850?lang=en) for supplying the data I played around with.
>
> [image]
>
> **Please use the newer [...]

Doesn't seem to work：
[image]

WeilongXia, 2023-07-03:
> Doesn't seem to work： [image]↳

You should modify `livox_topic` to `"/livox/lidar"`, and `pointcloud_topic` to another one.
Hi author @kennyjchen , with `deskew` set to `"false"`, it can work with Livox-MID360 now as below!
[image]
 And the offset_time units in livox_ros_driver2 is microseconds.

WeilongXia, 2023-07-03:
Hi author, when i set `deskew` to `true`, the program is not working properly, this is the dataset i used: https://1drv.ms/u/s!AjkIEg1kWUNeiArwedvaRRYd5hWB?e=BaII3v
Could you please help to figure this out?

Joosoo1, 2023-07-04:
yes! DLIO is now working  with deskewing！just `livox_topic` to "/livox/lidar"

kennyjchen, 2023-07-05:
Good to hear!

yijing2050, 2023-07-06:
DLIO is now working with deskewing in Ubuntu 18.04, but it seems not stable and the path is not smooth as fast_lio2. And the map will fly in some time later.
[image]

juliangaal, 2023-07-11:
@Joosoo1 what are the extrinsics of the dataset you provided?

Joosoo1, 2023-07-12:
> @Joosoo1 what are the extrinsics of the dataset you provided?
Please use this dataset, I tested it and it works great!：https://1drv.ms/u/s!AjkIEg1kWUNeiArwedvaRRYd5hWB?e=BaII3v
The extrinsics are as follows：
[image]

CharlieV5, 2023-07-17:
May I ask you guys what is the meaning of dlio/pointcloud/dekew in dlio.yaml? Does it mean that this program will do the deskewing for pointcloud, or this program will consider the input points has been deskewed so it will no do the deskewing? What will happen when dlio/pointcloud/dekew is set as false? Thank you!

> You should modify `livox_topic` to `"/livox/lidar"`, and `pointcloud_topic` to another one. Hi author @kennyjchen , with `deskew` set to `"false"`, it can work with Livox-MID360 now as below! [...]

kennyjchen, 2023-07-17:
It will undistort the point cloud when set to true.

CharlieV5, 2023-07-18:
> It will undistort the point cloud when set to true.

Thanks

closed 2023-07-18 by kennyjchen

85256638, 2024-09-15:
> > > @Joosoo1 what are the extrinsics of the dataset you provided?
> > > Please use this dataset, I tested it and it works great!：https://1drv.ms/u/s!AjkIEg1kWUNeiArwedvaRRYd5hWB?e=BaII3v
> > > The extrinsics are as follows：
> > > ![image](https://private-user-images.githubusercontent.com/65029285/252843712-ec769812-5991-40df-9d9a-93bb19877555.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjYxNDYwODgsIm5iZiI6MTcyNjE0NTc4OCwicGF0aCI6Ii82NTAyOTI4NS8yNTI4NDM3MTItZWM3Njk4MTItNTk5MS00MGRmLTlkOWEtOTNiYjE5ODc3NTU1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTEyVDEyNTYyOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ0NDgzZGJjNDkxNTU0MTliMmRiMDJlOGYyNWZlYTU3MGFkNGIzYzIwNTE1MDBlNjQyMDU4ZWVkY2FlMzE0YTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.6QeoMoWEww-brlixtBHsRrXNcP5rUnOUNKV_IaBvoss)
> 
> Hi, can you kindly let me know where to update those extrinsic values? I guess in the `dlio.yaml`? can you kindly paste yours please? I would like to know how to adapt Livox Horizon's extrinsic from the datasheet to the dlio.yaml configuration.
> 
> ```
> dlio:
> 
>   version: 1.1.1
> 
>   adaptive: true
> 
>   pointcloud:
>     deskew: true
>     voxelize: true
> 
>   imu:
>     calibration: true
>     intrinsics:
>       accel:
>         bias: [ 0.0, 0.0, 0.0 ]
>         sm:   [ 1.,  0.,  0.,
>                 0.,  1.,  0.,
>                 0.,  0.,  1. ]
>       gyro:
>         bias: [ 0.0, 0.0, 0.0 ]
> 
>   extrinsics:
>     baselink2imu:
>       t: [ 0.006253, -0.011775, 0.007645 ]
>       R: [ 1.,  0.,  0.,
>            0.,  1.,  0.,
>            0.,  0.,  1. ]
>     baselink2lidar:
>       t: [ 0.,  0.,  0. ]
>       R: [ 1.,  0.,  0.,
>            0.,  1.,  0.,
>            0.,  0.,  1. ]
> ```
> 
> LIvox Horizon extrinsic: ![image](https://private-user-images.githubusercontent.com/3270742/367205751-f3791ad1-8704-4bac-ac88-5a44abfaf8de.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY0MDk1OTYsIm5iZiI6MTcyNjQwOTI5NiwicGF0aCI6Ii8zMjcwNzQyLzM2NzIwNTc1MS1mMzc5MWFkMS04NzA0LTRiYWMtYWM4OC01YTQ0YWJmYWY4ZGUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDkxNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA5MTVUMTQwODE2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NzNlOWE2NWVjMzFiNzFhODA1NWRiZTk4ZDI5MWI5NDFiNDYxNjdjNmUyOWEyYTMxNjBkN2JiMTQ2ZDU4NTAyYSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.B1knzeKU-nONNHOmjkxD4zZFlX9qwuRDuD_Asq5zi1A)

@kennyjchen @Joosoo1 this issue has been closed, I would love to hear your advice :) thanks!

thisisdipanjan, 2026-03-02:
> May I ask you guys what is the meaning of dlio/pointcloud/dekew in dlio.yaml? Does it mean that this program will do the deskewing for pointcloud, or this program will consider the input points has [...]

The pointcloud topic has to be livox/lidar right so whats the livox topic is about? can it stay livox/lidar ?