vectr-ucla/direct_lidar_inertial_odometry #7: keyframe-based loop closure in paper

michael-sz, 2023-06-18:
great job! thanks for sharing!

1. when i read the paper, i found the part of loop closure, but not found during the code .  will it be released in future? 
2.  if add gps positon as prior factor,  the way of lio-sam based on gtsam will be better?

michael-sz, 2023-06-20:
got it, thanks for reply.
And if as you convenience , please show  me  the main distinction between dlio and dilom ,thanks.

kennyjchen, 2023-06-21:
tl;dr -- 

- DLIO: LIO with fast and accurate motion correction
- DLIOM: SLAM with innovations in keyframing (resistances to slippage), submapping (looks at overlap / Jaccard index), and mapping (connectivity between keyframes) to increase overall reliability

closed 2023-06-26 by kennyjchen