vectr-ucla/direct_lidar_inertial_odometry #20: Differences in GICP between DLIO and DLO

Bugday001, 2023-08-07:
Hi, 
Thank you for your work!
I found that a guess transformation matrix was passed in for the s2m gicp align function in DLO, while DLIO transforms each new scan pointcloud into global frame for GICP. Does this have any impact on the accuracy or efficiency of GICP solving? Also, are the GICP used by these two algorithms exactly the same?:D

kennyjchen, 2023-08-11:
Hi @Bugday001 -- Yes, DLIO's input scans already encode the initial guess from IMU propagation, so there's no need to pass a prior into the function (as DLO does). This doesn't (shouldn't) affect the accuracy / efficiency of GICP at all, since it's just a question of what reference frame the operation is being performed in. And yes, both DLO and DLIO use NanoGICP as its scan-matching optimizer.

kennyjchen, 2023-08-11:
Hi @Bugday001 -- Yes, DLIO's input scans already encode the initial guess from IMU propagation, so there's no need to pass a prior into the function (as DLO does). The optimization result could be slightly different---not because of the difference in reference frame, but rather since DLO uses s2s as its prior to s2m while DLIO uses full IMU propagation as its prior to s2m---but we haven't see any issues with this. But yes, both DLO and DLIO use NanoGICP as its scan-matching optimizer, so the algorithms are the same.

Bugday001, 2023-08-14:
Thanks for your reply, I will close this issue.

closed 2023-08-14 by Bugday001