vectr-ucla/direct_lidar_inertial_odometry #23: Struggling with Point Cloud Distortion Correction while Adapting HESAI (Pandar) LiDAR

JokerJohn, 2023-09-24:
Hello, Kenny!

Thank you for your remarkable work. I am currently working on adapting HESAI (Pandar) LiDAR and encountering difficulties progressing with the point cloud distortion correction. Can you please provide some guidance or suggestions on how to resolve this issue?

Any help would be greatly appreciated.

[image]

JaySlamer, 2023-11-03:
> Hello, Kenny!
> 
> Thank you for your remarkable work. I am currently working on adapting HESAI (Pandar) LiDAR and encountering difficulties progressing with the point cloud distortion correction. Can you please provide some guidance or suggestions on how to resolve this issue?
> 
> [image] The terminal log is as follows. [image] I have added some log statements to pinpoint the problem, and it seems like the LIDAR timestamp is less than the latest timestamp in the IMU buffer. My IMU is a 6-axis one operating at 100Hz, and I understand the characteristics of the HESAI point cloud timestamp. Regarding the parameters, I have only modified the extrinsic parameters and the related topics.
> 
> Could you please provide some insights or suggestions on how to address this issue? Any help would be greatly appreciated.
I am also using HESAI lidar and have no problem with it. Here are some of my tips:
1. In the author's original code,timestamps need to be absolute instead of relative to the first point. 
if your timestamps are relative:
            extract_point_time = [&sweep_ref_time](boost::range::index_value<PointType&, long> pt)
            { return sweep_ref_time + pt.value().timestamp; };
if they are absolute:
            extract_point_time = [&sweep_ref_time](boost::range::index_value<PointType&, long> pt)
            { return pt.value().timestamp; };
3. make sure the unit(second/millisecond/microsecond/nanosecond) is correct
4. make sure the sensors are synchronized.

JokerJohn, 2023-11-07:
@JaySlamer Thank you for your kind response. I've identified the issue at hand. The timestamp in my lidar point cloud message represents the end of a point cloud scan, not the beginning. Additionally, the timestamp for the first point in the cloud is incorrect and does not match the message timestamp. I will process the data accordingly and give it another try. Thank you for your patient assistance.

closed 2023-11-07 by JokerJohn