vectr-ucla/direct_lidar_inertial_odometry #27: Intrinsics and extrinsics

ubicray, 2023-11-13:
Hey! Firstly, thanks for great paper and code!
I used this tool for intrinsics and extrinsics: https://github.com/hku-mars/LiDAR_IMU_Init
I need help on understanding how to implement those values for DLIO:

Initialization result:
Rotation LiDAR to IMU (degree)     = -0.084620 -0.649203 -1.151916
Translation LiDAR to IMU (meter)   = -0.056023 -0.021736  0.026451
Time Lag IMU to LiDAR (second)     = -0.010057
Bias of Gyroscope  (rad/s)         = -0.007985 -0.012948  0.020327
Bias of Accelerometer (meters/s^2) = 0.010086 0.009814 0.010098
Gravity in World Frame(meters/s^2) = -0.027753  0.030412 -9.809914

Homogeneous Transformation Matrix from LiDAR to IMU: 
 0.999734  0.020119 -0.011298 -0.056023
-0.020101  0.999797  0.001704 -0.021736
 0.011330 -0.001477  0.999935  0.026451
 0.000000  0.000000  0.000000  1.000000

Refinement result:
Rotation LiDAR to IMU (degree)     = -0.014112 -0.341810 -1.171677
Translation LiDAR to IMU (meter)   = -0.042382 -0.019593  0.020312
Time Lag IMU to LiDAR (second)     = -0.010057
Bias of Gyroscope  (rad/s)         = -0.008959  0.001204  0.013092
Bias of Accelerometer (meters/s^2) =  0.013954  0.009286 -0.020293
Gravity in World Frame(meters/s^2) = -0.029866  0.030184 -9.791937

Homogeneous Transformation Matrix from LiDAR to IMU: 
 0.999773  0.020448 -0.005959 -0.042382
-0.020446  0.999791  0.000368 -0.019593
 0.005965 -0.000246  0.999982  0.020312
 0.000000  0.000000  0.000000  1.000000

Btw, using livox mid-360

greymfm, 2024-07-22:
Homogeneous Transformation Matrix is defined as:
```
r11 r12 r13   t1
r21 r22 r23   t2
r31 r32 r33   t3
0    0    0      1
```
with R rotation matrix and T transformation vector
I would try something like this:

  imu:
    calibration: false
    intrinsics:
      accel:
        bias: [  0.013954, 0.009286, -0.020293 ]
        sm:   [ 1.,  0.,  0.,
                0.,  1.,  0.,
                0.,  0.,  1. ]
      gyro:
        bias: [  -0.008959, 0.001204, 0.013092 ]

  extrinsics:
    baselink2imu:
      t: [ -0.042382, -0.019593, 0.020312 ]
      R: [ 1.,  0.020448,  0.,
           -0.020446,  1.,  0.,
           0.,  0.,  1. ]
    baselink2lidar:
      t: [ 0.,  0.,  0. ]
      R: [ 1.,  0.,  0.,
           0.,  1.,  0.,
           0.,  0.,  1. ]

85256638, 2024-09-02:
> Homogeneous Transformation Matrix is defined as:
> 
> ```
> r11 r12 r13   t1
> r21 r22 r23   t2
> r31 r32 r33   t3
> 0   0   0     1
> ```
> 
> with R rotation matrix and t transformation vector Based on your calibration values, I would try something like this:
> 
> ```
> 
>     imu:
>     calibration: false
>     intrinsics:
>       accel:
>         bias: [  0.013954, 0.009286, -0.020293 ]
>         sm:   [ 1.,  0.,  0.,
>                 0.,  1.,  0.,
>                 0.,  0.,  1. ]
>       gyro:
>         bias: [  -0.008959, 0.001204, 0.013092 ]
> 
> 
>   extrinsics:
>     baselink2imu:
>       t: [ -0.042382, -0.019593,   0.020312 ]
>       R: [ 1.,         0.020448,   0.,
>            -0.020446,  1.,         0.,
>            0.,         0.,         1. ]
>     baselink2lidar:
>       t: [ 0.,  0.,  0. ]
>       R: [ 1.,  0.,  0.,
>            0.,  1.,  0.,
>            0.,  0.,  1. ]
> ```
> 
> Maybe you can give some feedback what parameters you actually did use for the Livox MID-360 (what did work well for you)? ;-)

since the OP didn't report back on the above setting. I would like to share my settings:
I'm using Livox Avia, the intrinsics and extrinsics are from: https://github.com/hku-mars/LiDAR_IMU_Init as below:

```
Initialization result:
Rotation LiDAR to IMU (degree)     = -1.313296 -1.126729 -0.301456
Translation LiDAR to IMU (meter)   =  0.016343  0.055562 -0.074086
Time Lag IMU to LiDAR (second)     = -0.003865
Bias of Gyroscope  (rad/s)         =  0.003193 -0.000242  0.003296
Bias of Accelerometer (meters/s^2) = -0.009860 -0.010175 -0.009963
Gravity in World Frame(meters/s^2) = -0.020684  0.188709 -9.808163

Homogeneous Transformation Matrix from LiDAR to IMU: 
 0.999793  0.005710 -0.019536  0.016343
-0.005260  0.999721  0.023021  0.055562
 0.019662 -0.022913  0.999544 -0.074086
 0.000000  0.000000  0.000000  1.000000

Refinement result:
Rotation LiDAR to IMU (degree)     = -1.321007 -0.742159  0.078993
Translation LiDAR to IMU (meter)   =  0.049017  0.023597 -0.070867
Time Lag IMU to LiDAR (second)     = -0.003865
Bias of Gyroscope  (rad/s)         = 0.002492 0.001597 0.000897
Bias of Accelerometer (meters/s^2) =  0.004518 -0.006717 -0.014002
Gravity in World Frame(meters/s^2) = -0.011651  0.187335 -9.802430

Homogeneous Transformation Matrix from LiDAR to IMU: 
 0.999915 -0.001080 -0.012980  0.049017
 0.001378  0.999734  0.023034  0.023597
 0.012952 -0.023050  0.999650 -0.070867
 0.000000  0.000000  0.000000  1.000000
```
I applied the above intrinsics and extrinsics into the **dlio.yaml**:

```
dlio:

  version: 1.1.1

  adaptive: true

  pointcloud:
    deskew: true
    voxelize: true

  imu:
    calibration: false
    intrinsics:
      accel:
        bias: [ 0.004518, -0.006717, -0.014002 ]
        sm:   [ 1.,  0.,  0.,
                0.,  1.,  0.,
                0.,  0.,  1. ]
      gyro:
        bias: [ 0.002492, 0.001597, 0.000897 ]

  extrinsics:
    baselink2imu:
      t: [ 0.049017, 0.023597, -0.070867 ]
      R: [ 0.999915,  -0.001080,  -0.012980,
           0.001378,  0.999734,  0.023034,
           0.012952,  -0.023050,  0.999650 ]
    baselink2lidar:
      t: [ 0.,  0.,  0. ]
      R: [ 1.,  0.,  0.,
           0.,  1.,  0.,
           0.,  0.,  1. ]
```

the result is rather shocking:

[image]

I only moved less than 5 meters, but the

```
Distance Traveled:  291.49 meters.
Distance to Origin: 181.76 meters.
```
I used the "rosservice call /robot/dlio_map/save_pcd 0.01 /home/test/ws/src/direct_lidar_inertial_odometry/PCD" to save the PCD file and it looks very off.

85256638, 2024-09-15:
@kennyjchen I would be delighted if you could give some advice :)

kennyjchen, 2024-09-16:
@85256638 The extrinsics seem relatively correct (looking at [this](https://github.com/vectr-ucla/direct_lidar_inertial_odometry/issues/5#issuecomment-2348306887). My guess is that it's an issue with the FOV of the Aria. DLIO's keyframing scheme really relies on the 360 nature of mechanical LiDARs. It should work for Livox Mid-360's though (I've tested that).

greymfm, 2025-07-06:
So, in my latest simulations (with simulated Livox MID360, LiDAR rotated and translated from base_link), I did write a python script to compute the rotation extrinsics (baselink to lidar, baselink to imu):

[image]

Also, I commented out the transformation publishing  (transform: baselink to lidar and transform: baselink to imu) in the DLIO code as my simulation already does this. 

Seems to work (I assume this is the correct way to do this with rotated/translated LiDAR positions?).

file:///home/alex/Pictures/Screenshots/Screenshot%20from%202025-07-06%2007-29-50.png
```
import json
import numpy as np
from scipy.spatial.transform import Rotation as R

# Drehungen in Radiant (first roll, then pitch, finally yaw)
roll = -0.4                 # X-Achse
pitch = 0   # np.deg2rad(25)   # Y-Achse
yaw =   -1.57  #  np.deg2rad(0)     # Z-Achse

print('roll,pitch,yaw (rad):', roll, pitch, yaw)
print()

# Rotation von base_link zu lidar
r = R.from_euler('xyz', [roll, pitch, yaw])
ROT = r.as_matrix()
ROT = np.round(ROT, 6)

print(ROT.tolist())

R_flat = (ROT.flatten()).tolist()

# In JSON schreiben
json_data = json.dumps({"R": R_flat}, indent=2)
print(json_data)

```