vectr-ucla/direct_lidar_inertial_odometry #35: runtime issues when compiling with clang

JanQuenzel, 2024-02-01:
Hey there,
when I compile dlio (and dlo) with clang, the average computation time more than triples in comparison to using gcc ( ~100ms instead of ~30ms for an Ouster OS0 128).
I tried this on a AMD Ryzen 7 5800H with ubuntu 20.04 using clang 12.0 and gcc 9.4.
With clang, the CPU was running at full core utilization (x16) and a CPU load around 99% while dropping around 30% of the scans.
With gcc, the CPU runs at around 8% and slightly above 1 CPU utilized.
Any idea were this problem may come from?

Best regards, Jan

kennyjchen, 2024-05-16:
Hmm, probably has something to do with OpenMP when compiling with gcc.

SnifferCaptain, 2026-02-26:
if(CMAKE_CXX_COMPILER_ID STREQUAL "Clang")
        set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -pthread -fopenmp=libgomp")
        set(CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} -lgomp")# eigen conflict with omp in clang
    else()
        set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -pthread -fopenmp")
    endif()