vectr-ucla/direct_lidar_inertial_odometry #44: downloading PCD files

yahyahajlaoui, 2024-05-23:
First, Thank you for this amazing work.
Please I didn't understand the leaf_size var in the rosservice command that download the pcd file, 
If I want to save the dense map, what is the number of leaf_size to put in the command ? I tried 0.01 and the pcd file downloaded isn't like the dense map in rviz I don't know why.
Thank you again.

85256638, 2024-09-01:
> The dense map in rviz is purely for visualization and is just the deskewed scan but with a large decay time and an adjusted alpha value to give it that sharp look. The actual map used internally consists of the keyframes used for localization, and that is what is saved with the service command.

thanks, so what leaf_size would you recommend to use?

Andy-Leo10, 2025-11-08:
Excuse me, did you have any problems saving the map? The service doesn't work for me when I call it.

Following the link, you will see the steps I followed. I used ROS2 Humble.
https://github.com/vectr-ucla/direct_lidar_inertial_odometry/issues/86#issuecomment-3476967755