vectr-ucla/direct_lidar_inertial_odometry #49: `baselink2imu` and `baselink2lidar` are reversed

VRichardJP, 2024-06-04:
Despites their name, both `baselink2imu` and `baselink2lidar` are used as imu->baselink and lidar->baselink transformations in the code.

For example, this line:

https://github.com/vectr-ucla/direct_lidar_inertial_odometry/blob/77b356325dd629cc6f0f1a048bd867ca5ff3d5c3/src/dlio/odom.cc#L1386

converts IMU angular velocity measurement into baselink frame.

Another example:

https://github.com/vectr-ucla/direct_lidar_inertial_odometry/blob/77b356325dd629cc6f0f1a048bd867ca5ff3d5c3/src/dlio/odom.cc#L575-L576

Here, the original LIDAR scan (so in lidar frame) is first transformed by `baselink2lidar`, as if it was "lidar->baselink".

Not only it makes the code confusing, but it is also easy to miswrite the transforms in the `dlio.yaml` file. I think it would be better to rename these transforms to `lidar2baselink` and `imu2baselink`:

```bash
$ grep -rIl baselink2lidar | xargs sed -i 's/baselink2lidar/lidar2baselink/g'
$ grep -rIl baselink2imu | xargs sed -i 's/baselink2imu/imu2baselink/g'
```

?

kennyjchen, 2024-06-05:
Good point -- technically yes they represent the transformation from sensor to baselink. Sorry for the confusion.

iampaulidrobo, 2025-08-25:
Is the issue closed?