vectr-ucla/direct_lidar_inertial_odometry #70: base PCD map location?

lonlonago, 2024-09-06:
If we have a PCD map, can we use this code to make a location based on the PCD map?

 If possible, how do we need to modify this code

kennyjchen, 2024-09-11:
You can essentially edit the code so that it loads in a PCD, sets it as the first (and only) keyframe, and disables tracking of subsequent keyframes.

lonlonago, 2024-09-12:
Thank you very much for your reply!
Yes, I did, but the odometer kept jumping and reported an error after running for a short period of time. The error message:

Terminate called after throwing an instance of 'ros::serialization::StreamOverrunException'

here is my code change:
`void dlio::OdomNode::buildSubmap(State vehicle_state) {

    //   submap to whole map test 

        pcl::PointCloud<PointType>::Ptr tmp_map_cloud (boost::make_shared<pcl::PointCloud<PointType>>());

    pcl::io::loadPCDFile("/home/lan/ouster_zijin.pcd", *tmp_map_cloud);
        this->gicp_temp.setInputTarget(tmp_map_cloud);

        this->submap_kdtree = this->gicp_temp.target_kdtree_;

    this->submap_cloud = tmp_map_cloud;
    // this->submap_normals = this->gicp_temp.getSourceCovariances;  // getTargetCovariances
    this->submap_normals = this->gicp_temp.getTargetCovariances();  // getTargetCovariances

    this->submap_kf_idx_prev = this->submap_kf_idx_curr;

}`

only build submap use once here 

`  // Set initial frame as first keyframe
  if (this->keyframes.size() == 0) {
    this->initializeInputTarget();
    this->main_loop_running = false;
    this->submap_future =
      std::async( std::launch::async, &dlio::OdomNode::buildKeyframesAndSubmap, this, this->state );
        this->buildSubmap(this->state);  //  only build submap use once here 

    this->submap_future.wait(); // wait until completion
    return;
  }
`

Thank you very much for taking the time to review this code and problem