vectr-ucla/direct_lidar_inertial_odometry #71: There is a drift on z-axis when run ros2 branch with Livox mid360

hoteaaa, 2024-09-19:
Thank you for open-sourcing your excellent project!However there are some problems when I run the ROS2 branch.

I kept the lidar still and started the program directly, I noticed that the Position of the z-axis had a big drift and the Accel Bias stayed at -5 on the z-axis. in addition, the computation time stays 0. When I move the radar, the data gets more anomalous.

[image]

There was a similar closed issue [(issue_about_livox_mid360)](https://github.com/vectr-ucla/direct_lidar_inertial_odometry/issues/5) and I tried modifying the extrinsics/baselink2imu/t and changing the code to fit the units of the livox's imu, but neither of them worked.

Here is my build and run process, please point out any errors. Looking forward to your reply.

BUILD
`cd ws/src`
`git clone https://github.com/vectr-ucla/direct_lidar_inertial_odometry -b feature/ros2`
`cd ..`
`colcon build --symlink-install --packages-select direct_lidar_inertial_odometry`

RUN
`ros2 launch livox_ros_driver2 rviz_MID360_launch.py `
`ros2 launch direct_lidar_inertial_odometry dlio.launch.py rviz:=true pointcloud_topic:=livox/lidar imu_topic:=livox/imu`

closed 2024-09-26 by hoteaaa

85256638, 2024-09-27:
can you provide your solution before closing down the issue? I would help people who might have the same issue.

kennyjchen, 2024-09-27:
Does this happen on ROS1? Is there a dataset I can test with?

hoteaaa, 2024-10-04:
> can you provide your solution before closing down the issue? I would help people who might have the same issue.

modify the code in the file "odom.cc":
`  Eigen::Vector3f lin_accel(imu_raw->linear_acceleration.x,
                            imu_raw->linear_acceleration.y,
                            imu_raw->linear_acceleration.z);`
to
`  Eigen::Vector3f lin_accel(imu_raw->linear_acceleration.x * this->gravity_,
                            imu_raw->linear_acceleration.y * this->gravity_,
                            imu_raw->linear_acceleration.z * (-this->gravity_ ));`

hoteaaa, 2024-10-04:
> Does this happen on ROS1? Is there a dataset I can test with?

sry,i only test it on ROS2 humble

hoteaaa, 2024-10-04:
> can you provide your solution before closing down the issue? I would help people who might have the same issue.

By the way,about the computation time,i guess the problem is the clock.I use the clock of ROS2 instead and solve the problem.

hoteaaa, 2024-10-04:
@85256638 
1.I test on ros2 branch on ROS2 humble. i find the origin extrinsics/baselink2imu/R is [-1,0,0; 0,1,0;0,0,-1].it's wrong with mid-360;
i modify it,and this is my dlio.yaml.
```/**:
  ros__parameters:

    version: 1.1.1

    adaptive: false

    pointcloud/deskew: true
    pointcloud/voxelize: true

    imu/calibration: true
    imu/intrinsics/accel/bias: [  0., 0., 0. ]

    imu/intrinsics/accel/sm:   [ 1.,  0.,  0.,
                                 0.,  1.,  0.,
                                 0.,  0.,  1. ]
    imu/intrinsics/gyro/bias: [ 0., 0., 0. ]

    extrinsics/baselink2imu/t: [  -0.042382, -0.019593,   0.020312 ] #-0.011, -0.02329, 0.04412
    extrinsics/baselink2imu/R: [  1.,  0.,  0.,
                                  0.,  1.,  0.,
                                  0.,  0.,  1. ]
    extrinsics/baselink2lidar/t: [ 0.,  0.,  0., ]
    extrinsics/baselink2lidar/R: [ 1.,  0.,  0.,
                                   0.,  1.,  0.,
                                   0.,  0.,  1. ]```
2.modify the code in file 'odom.cc',about line 1400:
`  Eigen::Vector3f lin_accel(imu_raw->linear_acceleration.x * this->gravity_,
                            imu_raw->linear_acceleration.y * this->gravity_,
                            imu_raw->linear_acceleration.z*(this->gravity_));`
3.About the computation time ,i guess it the problem of the clock. I use the clock class of ROS2  instead and it work!

Thanks for your excellent work!@kennyjchen